greatscottgadgets/libbtbb #13: Wireshark btbb plugin does not dissect latest file format

ikelos, 2015-02-18:
I've built libbtbb and ubertooth from git (352b420f and d09308f4) in order to be able to create pcaps directly from ubertooth without kismet-ubertooth.  I've successfully saved several of the pcaps/pcapngs but upon openning them in wireshark they're just reported as WTAP_ENCAP = 161, which seems to be correctly interpretted as "Bluetooth BR/EDR Baseband RF (161)".  However, there's no further dissection.

Opening the example pcap file works fine, and there are no errors, so I suspect the dissector is working fine, but is for an older file format (as mentioned in issue #9).  
"Yes, those plugins are old. We need to mostly rewrite them for the new file format." (July 2014)

It's rather confusing that the package has been updated to work with wireshark-1.12, but will not read files generated by the latest version of the library itself.  Could this please be documented if not actually fixed?

dominicgs, 2015-02-18:
The Ubertooth tools are capable of writing the new format pcap/pcapng files, but I need to write the dissector for Wireshark (for "Bluetooth BR/EDR Baseband RF (161)").  It is the last remaining thing that I need to do before the next release (which I was hoping would be this month).

If capturing BLE data, the dissector should already be included with Wireshark 1.12.

1am, 2015-02-18:
@ikelos @dominicgs So if I understand correctly, one way to work around this issue before next release would be using an older version of Wireshark?

dominicgs, 2015-02-18:
No, there is no dissector for the new format yet.  The work around is to use the old format with the old dissector until I have finished it.

For Basic Rate Bluetooth sniffing only Kismet outputs the old file format.

1am, 2015-02-18:
@dominicgs How do I get to choose the format?

dominicgs, 2015-02-18:
By using the previous release.  Git is in the process of transitioning from the old format to the new.

1am, 2015-02-18:
@dominicgs I cleaned up after checkout from libbtbb master branch, got back to libbtbb-2014-02-R4 downloaded 2 days ago and reinstalled it along with btbb.so plugin for Wireshark and ubertooth host utils.
I still get the the same PPI packages. Which other release should I try? Maybe now there's an issue with my firmware (**git-c9d3ed7**) being too new?  
In the **libbtbb-2014-02-R4/wireshark/plugins(* directory there are also the following directories:
**btatt**, **btbb**, **btle**, **btsm**. Should try installing all of them?

dominicgs, 2015-02-18:
@1am This issue is unrelated to your PPI query.  This is about "Bluetooth BR/EDR Baseband RF (161)" packet dissectors, not BLE.

1am, 2015-02-18:
Ah, ok.

straithe, 2021-02-17:
I'm going to close this for now as there hasn't been a response in a while and it looks like there are no further questions. Please re-open this issue or open a new one if you still need assistance.

closed 2021-02-17 by straithe