greatscottgadgets/libbtbb #16: wmem missing when building wireshark plugin btbb

rethab, 2015-03-31:
Hi, I've tried building different versions of the btbb wireshark plugin against different versions of wireshark, but it's the same in all cases:
epan/wmem/wmem.h: No such file or directory

This file is included from /usr/include/wireshark/epan/proto.h. I understand this is not part of libbtbb, but given that wireshark did install (from source) with no issues leads me to think you might be able to give me a hint.

This guy seems to have the same problem, but I couldn't see an answer there: http://permalink.gmane.org/gmane.os.apple.macports.user/33078 Note that I also see the wmem.h in the source tree, but it doesn't end up in /usr/include.. 

In terms of versions, the latest test I did with wireshark 1.11.3 and libbtbb master I also tried libbtbb 2014-04-R2.

Thanks a lot!

dominicgs, 2015-04-01:
Hi,

Wireshark doesn't install header files when built from source, so you must tell cmake where to find them.  Use ```cmake -DWIRESHARK_INCLUDE_DIRS=/path/to/headers .```

rethab, 2015-04-01:
Interesting. I thought the -dev version (arch linux in my case) would install those as well, but that's apparently not the case.

It works with the headers from the source tree though. Thanks a lot!

closed 2015-04-01 by rethab