greatscottgadgets/libbtbb #21: Wireshare plugin finding malformed packets only

epablosensei, 2015-08-03:
I setup my ubertooth one with kismet and all the libraries by following in the setup.

Kismet seems to be seeing the traffic, at least the packet count goes up, but when I try to read them in wireshark all I see are Malformed Packets.  Also the destination is always set to 00:00:00_00:00:00 
See the attached screen shots.

[image]

Any ideas?

dominicgs, 2015-08-03:
Could you tell us which version of libbtbb, ubertooth and kismet you are using?  Also, could you tell us a little about the devices that are transmitting?

ID packets use the address of 9E:8B:33, so if we're making the assumption that they are malformed ID packets, we are assuming that there are 15 bit errors.  I think that is unlikely.

What is more likely to be happening is that those are LAPs and no UAP has been found yet, so they packets cannot be interpreted, this limiting them to 14 bytes.  We are then making the (incorrect) assumption that a 14 byte packet is an ID packet as they are the only valid 14 byte packets.

The destination address will almost always be unknown for Bluetooth devices due to the lack of destination address in the packet.  There are possibly circumstances where we could work it out, but they are not possible yet.

epablosensei, 2015-08-03:
Hi Dominic,

thanks for the quick response.

Versions are as follows:
* Kismet: 2013-03-R1b
* libbtbb: git head ac7dabc 2015-06-12
* ubertooth: Firmware revision: 2014-04-R1

I've been running my tests with different devices:
* 9B:72:C8 - Nexus 7 which is sending music to a bluetooth speaker
* 5C:F3:70:67:F9:56 and 5C:F3:70:67:F9:2A are Broadcom Corp. BCM20702A0 Bluetooth 4.0 from which I'm doing some bluetooth test on RFCOMM layer using python -> these are actually what I want to be able to sniff

dominicgs, 2015-08-03:
The new dissectors and packet format in git are in a very early and experimental stage for Basic Rate sniffing.

I would suggest that you roll back to the 2014-02-R1 release of libbtbb and Ubertooth host code.

epablosensei, 2015-08-03:
So I downgraded everything 2014-02-R2, because I could not find R1.

Nothing really better.
Actually now ubertooth-specan-ui failes with:
```
Traceback (most recent call last):
  File "/usr/local/bin/ubertooth-specan-ui", line 260, in <module>
    window = Window()
  File "/usr/local/bin/ubertooth-specan-ui", line 227, in __init__
    self._device = self._open_device()
  File "/usr/local/bin/ubertooth-specan-ui", line 250, in _open_device
    return Ubertooth(device)
TypeError: __init__() takes exactly 1 argument (2 given)
```

dominicgs, 2015-08-03:
What does "nothing really better mean"?  What has changed? What is the same? What are you seeing in wireshark? 

Have you completely removed the other copy of the specan-ui? It looks like both versions are installed and conflicting.

epablosensei, 2015-08-03:
OK.. so let me correct.
Just to be sure, after you suggestion I clean up /usr/local/ with an `rm -rf /usr/local/*`

Rebuilt and re-installed libbtbb and the Ubertooth tools using 2014-02-R2, Kismet 2013-03-R1b

Wireshark pcap looks just the same (all malformed packets) and ubertooth-specan-ui is also failing in the same way.

epablosensei, 2015-08-04:
OK.. So I decided to redo everything on a new ubuntu 12.04 VM.
Built everything with 2014-02-R1.

* ubertooth-specan-ui  works here
* kismet seams to work
* I can't get the wireshark plugin to compile. I'm using wireshark 1.12.4-1ubuntu1~ppa2~p

```
[ 25%] Building C object CMakeFiles/btbb.dir/packet-btbb.c.o
In file included from /usr/include/wireshark/epan/proto.h:46:0,
                 from /usr/include/wireshark/epan/packet.h:26,
                 from /home/epablo/git/bluetooth/libbtbb-2014-02-R1/wireshark/plugins/btbb/packet-btbb.c:32:
/usr/include/wireshark/epan/emem.h:29:30: fatal error: ws_symbol_export.h: No such file or directory
compilation terminated.
make[2]: *** [CMakeFiles/btbb.dir/packet-btbb.c.o] Error 1
make[1]: *** [CMakeFiles/btbb.dir/all] Error 2
make: *** [all] Error 2
```

any ideas?

epablosensei, 2015-08-04:
I'm starting to think that the problem may be in the wireshark plugin and not in the capturing.
Could you take a look at these files, maybe you can open them correctly on your machine.

https://vm1.rootdojo.com/index.php/s/untMHkVUtCTRmH0

dominicgs, 2015-08-05:
Did you manage to get the 2014-02-R1 packet dissector to build correctly?  How did you resolve the error finding ```ws_symbol_export.h```?

As long as your host is unable to calculate the UAP and CLK of the remote Bluetooth piconet, it will fail to decode any more than the access code of a packet.  I believe that this is what you are seeing when it refers to malformed packets.

P.S. Sorry for the delayed response, I'm out of the office until the 18th and won't have constant internet access.

epablosensei, 2015-08-05:
The 2014-02-R1 packet dissector (wireshark plugin) is still giving me the
same error.
I just used the plugin I had compile from 2014-02-R2 on my other machine.

When I see the kismet server logs, it does look like I has the information
(UAP and CLK), but I'm going to have to check the details.

Saludos cordiales / Best regards / Mit freundlichen Grüßen,
Pablo Endres

---
Pablo Endres <epablo@PabloEndres.com>
Web: http://www.PabloEndres.com
Mobile: +49 151 521 75909

epablosensei, 2015-08-18:
Hi Dominic,

so after lots of tests, I reinstalled my machine and I'm working on a clean Fedora 22 install. 
I rebuild everything using 2014-02-R2, kismet-2013-03-R1b and wireshark 1.12.6

Everything seems to have built ok (a couple of warnings, but nothing out of the ordinary). Kismet seems to be working well, but I have some problems with the wireshark plugin:

```
Couldn't load module /usr/lib64/wireshark/plugins/btbb.so: /usr/lib64/wireshark/plugins/btbb.so: undefined symbol: check_col
```

Any suggestions on how to get this working?

epablosensei, 2015-08-19:
So,

I think the wireshark error is related to this warning when I build the plugin:

```
/home/user/git/bluetooth/libbtbb-2014-02-R2/wireshark/plugins/btbb/packet-btlmp.c: In function ‘dissect_btlmp’:
/home/user/git/bluetooth/libbtbb-2014-02-R2/wireshark/plugins/btbb/packet-btlmp.c:1888:6: warning: implicit declaration of function ‘check_col’ [-Wimplicit-function-declaration]
  if (check_col(pinfo->cinfo, COL_PROTOCOL))
```

dominicgs, 2015-08-19:
This is a problem with the plugin and Wireshark 1.12.  You can find a fixed betsion of the btbb plugin in git.  You do not need to install libbtbb, just the wireshark plugin from the git repository.  The only change is the removal of check_col

mikeryan, 2016-02-04:
Closing this issue because it appears to be resolved. Please reopen if that is not the case.

closed 2016-02-04 by mikeryan