greatscottgadgets/libbtbb #22: Build error bluetooth_rxtx

SergeyPoltavtsev, 2015-08-13:
Hi,

I am trying to build firmware manually because of this issue (https://github.com/greatscottgadgets/ubertooth/issues/55) and I am getting following errors:

/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: cannot find -lstdc++
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: cannot find -lsupc++
collect2: error: ld returned 1 exit status
make: *** [bluetooth_rxtx.elf] Error 1

Any suggestions?

mikeryan, 2015-08-13:
This is not a libbtbb issue, and this is thus the incorrect tracker to post it in.

That said, are you attempting to build bluetooth_rxtx directly or are you building the entire firmware tree? You should be able to build just bluetooth_rxtx and that should not require libstdc++

SergeyPoltavtsev, 2015-08-14:
I am trying to build to build only bluetooth_rxtx. I successfully build it on mac but it didn`t solve problem with freezing. And still can`t build it on ubuntu (14.04 32bit)

SergeyPoltavtsev, 2015-08-14:
Finally, solved problem with hold using patch in this thread https://github.com/greatscottgadgets/ubertooth/issues/15

closed 2015-08-17 by SergeyPoltavtsev