greatscottgadgets/libbtbb #24: cannot build from git with odd error

ZeroChaos-, 2015-08-25:
[32;01m * [39;49;00mPackage:    net-libs/libbtbb-9999
[32;01m * [39;49;00mRepository: maintainer-zero
[32;01m * [39;49;00mMaintainer: zerochaos@gentoo.org
[32;01m * [39;49;00mUSE:        abi_x86_64 amd64 elibc_glibc kernel_linux pcap userland_GNU wireshark-plugins
[32;01m * [39;49;00mFEATURES:   preserve-libs sandbox userpriv usersandbox
>>> Unpacking source...
 [32;01m*[0m Fetching [1mhttps://github.com/greatscottgadgets/libbtbb.git[22m ...
git fetch https://github.com/greatscottgadgets/libbtbb.git +HEAD:refs/git-r3/HEAD
git symbolic-ref refs/git-r3/net-libs/libbtbb/0/__main__ refs/heads/master
 [32;01m*[0m Checking out [1mhttps://github.com/greatscottgadgets/libbtbb.git[22m to [1m/var/tmp/portage/net-libs/libbtbb-9999/work/libbtbb-9999[22m ...
git checkout --quiet master
GIT update -->
   repository:               https://github.com/greatscottgadgets/libbtbb.git
   at the commit:            389f0c690cd5e59bd0496ff06fec40711e2b0754
>>> Source unpacked in /var/tmp/portage/net-libs/libbtbb-9999/work
>>> Preparing source in /var/tmp/portage/net-libs/libbtbb-9999/work/libbtbb-9999 ...
>>> Source prepared.
>>> Configuring source in /var/tmp/portage/net-libs/libbtbb-9999/work/libbtbb-9999 ...
>>> Working in BUILD_DIR: "/var/tmp/portage/net-libs/libbtbb-9999/work/libbtbb-9999_build"
cmake --no-warn-unused-cli -C /var/tmp/portage/net-libs/libbtbb-9999/work/libbtbb-9999_build/gentoo_common_config.cmake -G Unix Makefiles -DCMAKE_INSTALL_PREFIX=/usr -DDISABLE_PYTHON=true -DPACKAGE_MANAGER=true -DPCAPDUMP=ON -DUSE_PCAP=ON -DCMAKE_BUILD_TYPE=Gentoo -DCMAKE_INSTALL_DO_STRIP=OFF -DCMAKE_USER_MAKE_RULES_OVERRIDE=/var/tmp/portage/net-libs/libbtbb-9999/work/libbtbb-9999_build/gentoo_rules.cmake -DCMAKE_TOOLCHAIN_FILE=/var/tmp/portage/net-libs/libbtbb-9999/work/libbtbb-9999_build/gentoo_toolchain.cmake  /var/tmp/portage/net-libs/libbtbb-9999/work/libbtbb-9999
Not searching for unused variables given on the command line.
loading initial cache file /var/tmp/portage/net-libs/libbtbb-9999/work/libbtbb-9999_build/gentoo_common_config.cmake
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/x86_64-pc-linux-gnu-gcc
-- Check for working C compiler: /usr/bin/x86_64-pc-linux-gnu-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/x86_64-pc-linux-gnu-g++
-- Check for working CXX compiler: /usr/bin/x86_64-pc-linux-gnu-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Pcap include dirs set to /usr/include
-- Pcap library set to  /usr/lib64/libpcap.so
-- Looking for pcap_version
-- Looking for pcap_version - found
-- Looking for pcap_open_dead
-- Looking for pcap_open_dead - found
-- Looking for pcap_freecode
-- Looking for pcap_freecode - found
-- Looking for pcap_breakloop
-- Looking for pcap_breakloop - found
-- Looking for pcap_create
-- Looking for pcap_create - found
-- Looking for pcap_datalink_name_to_val
-- Looking for pcap_datalink_name_to_val - found
-- Looking for pcap_datalink_val_to_description
-- Looking for pcap_datalink_val_to_description - found
-- Looking for pcap_datalink_val_to_name
-- Looking for pcap_datalink_val_to_name - found
-- Looking for pcap_findalldevs
-- Looking for pcap_findalldevs - found
-- Looking for pcap_free_datalinks
-- Looking for pcap_free_datalinks - found
-- Looking for pcap_get_selectable_fd
-- Looking for pcap_get_selectable_fd - found
-- Looking for pcap_lib_version
-- Looking for pcap_lib_version - found
-- Looking for pcap_list_datalinks
-- Looking for pcap_list_datalinks - found
-- Looking for pcap_set_datalink
-- Looking for pcap_set_datalink - found
-- Looking for pcap_open
-- Looking for pcap_open - not found
-- Looking for pcap_findalldevs_ex
-- Looking for pcap_findalldevs_ex - not found
[... 151 lines cut ...]
#endif

#include <gmodule.h>

#include "moduleinfo.h"

#ifndef ENABLE_STATIC
G_MODULE_EXPORT const gchar version[] = VERSION;

/* Start the functions we need for the plugin stuff */

G_MODULE_EXPORT void
plugin_register (void)
{
  {extern void proto_register_btbb (void); proto_register_btbb ();}
  {extern void proto_register_btlmp (void); proto_register_btlmp ();}
}

G_MODULE_EXPORT void
plugin_reg_handoff(void)
{
  {extern void proto_reg_handoff_btbb (void); proto_reg_handoff_btbb ();}
  {extern void proto_reg_handoff_btlmp (void); proto_reg_handoff_btlmp ();}
}
#endif

dominicgs, 2015-08-25:
It looks like we're using a member of the ```packet_info``` struct called ```private_data``` which doesn't exist in your version of the headers.

Which version of Wireshark are you building against?

dominicgs, 2015-08-25:
Looks like this commit is the culprit: https://code.wireshark.org/review/gitweb?p=wireshark.git;a=commit;h=6d207fe5f47f20f7c83d5c5da3e4ee0b0bca7b9b

ZeroChaos-, 2015-08-26:
sometimes building against wireshark-1.12.7
but mostly against wireshark-1.99.8

ZeroChaos-, 2015-08-26:
it's easy for me to build a different plugin based on the installed wireshark version.  so having a btbb-2 directory that build against wireshark 2 is an acceptable solution for me.

dominicgs, 2015-08-27:
I think I have a hacky fix for this, I'm not too concerned about a hacky fix in for this plugin as it will be obsolete soon.

dominicgs, 2016-01-21:
The difference between dissectors in Wireshark 1.x and 2.x is big enough that I decided to split them.  The 1.x plugins are now in `wireshark/plugins-legacy` and the 2.x plugins are in `wireshark/plugins`

Please test and let me know if you're happy with this fix.

ZeroChaos-, 2016-01-27:
both plugins were tested on amd64 and armv7-armv4.  please make a release that includes the 2.x plugin.

dominicgs, 2016-02-03:
Bug is fixed, this will be in the next release.

closed 2016-02-03 by dominicgs