greatscottgadgets/libbtbb #29: Conflict between libbtbb Wireshark plugins: btbb & btbredr

raulsiles, 2015-09-30:
The btbb and btbredr Wireshark plugins available in libbtbb 2015-09-R2 both have the same full name, "Bluetooth Baseband", and short name, "BT Baseband". Therefore, if both are built and installed, Wireshark cannot be started as there is a duplicated protocol name:

"Err  Duplicate protocol name "Bluetooth Baseband"! This might be caused by an inappropriate plugin or a development error."

The same applies to the Bluetooth Link Manager Protocol (LMP) provided by both plugins with the same name.

Which is the current and most recommended plugin, btbb or btbredr? 

Should we avoid reusing the same names to be able to have both enabled in Wireshark?

dominicgs, 2015-09-30:
@raulsiles Thanks for raising this issue.  It's known and there is a fix in git - https://github.com/greatscottgadgets/libbtbb/commit/ae840325b61ad74181b079db288b4309ce96746b

The btbredr plugin is newer but not fully tested, it is supported by the Ubertooth tools.  The btbb plugin is the format that the Kismet plugins produce.

raulsiles, 2015-09-30:
Thanks for the commit reference Dominic!

Although the Ubertooth Build Guide only details the building process for the libbtbb btbb and btbredr Wireshark plugins, is it recommended to also build the other plugins available in the current libbtbb 2015-09-R2 version: btatt, btle and btsm?

Or... should I assume that they are the same plugins/built-in dissectors available in current Wireshark versions, such as Wireshark 1.12.6 (included by default in Kali Linux 2.0.0)?

dominicgs, 2015-09-30:
The additional plugins are included in wireshark 1.12+, they are in the
release for anyone using earlier versions of wireshark.

raulsiles, 2015-09-30:
Thanks for the confirmation!

closed 2015-09-30 by raulsiles