greatscottgadgets/libbtbb #30: ubertooth-follow capture file unrecognizable  WTAP_ENCAP = 0

parkerpatriot, 2015-10-24:
I am trying to get ubertooth-follow to work and save the data to a pcapng file. I have installed the latest version of the ubertooth tools, updated libpcap, and also installed the latest release of libbtbb. Despite version notes that claim otherwise, I still get a `link-layer type 255 isn't supported in savefiles` when attempting to use the `-q` option with the `ubertooth-follow`. If I instead use the `-r` to capture to pcapng format wireshark fails to dissect the packet and just displays `WTAP_ENCAP = 0` in the Info column.

Version:
libbtbb: 2015-10-R1
ubertooth: 2015-10-R1
OS: Ubuntu 14   

[image]

straithe, 2021-05-04:
@parkerpatriot Are you still experiencing this issue?

straithe, 2021-06-08:
I'm going to close this as there hasn't been a response in the last month, but please re-open this issue or open a new one if you still need assistance.

closed 2021-06-08 by straithe