greatscottgadgets/libbtbb #33: btbb and btbredr wireshark plugin build error on Ubuntu 12.04 LTS

maximcherny, 2015-12-09:
Hi,

```
devel@ubuntu:~/dev/libbtbb-2015-10-R1/wireshark/plugins/btbb/build$ cmake -DCMAKE_INSTALL_LIBDIR=/usr/lib/wireshark/libwireshark1/plugins ..
-- The C compiler identification is GNU
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Plugin will be installed in: /usr/lib/wireshark/libwireshark1/plugins
-- checking for one of the modules 'glib-2.0>=2.4.0'
-- Found GLib2: glib-2.0 /usr/include/glib-2.0;/usr/lib/x86_64-linux-gnu/glib-2.0/include
-- Looking for include files HAVE_GLIB_GREGEX_H
-- Looking for include files HAVE_GLIB_GREGEX_H - found
-- Looking for include files HAVE_GLIB_GCHECKSUM_H
-- Looking for include files HAVE_GLIB_GCHECKSUM_H - found
-- Configuring done
-- Generating done
-- Build files have been written to: /home/devel/dev/libbtbb-2015-10-R1/wireshark/plugins/btbb/build
```

```
devel@ubuntu:~/dev/libbtbb-2015-10-R1/wireshark/plugins/btbb/build$ make
[ 25%] Building C object CMakeFiles/btbb.dir/packet-btbb.c.o
/home/devel/dev/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c: In function ‘dissect_fhs’:
/home/devel/dev/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c:212:14: warning: assignment makes pointer from integer without a cast [enabled by default]
/home/devel/dev/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c: At top level:
/home/devel/dev/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c:277:78: error: expected ‘;’, ‘,’ or ‘)’ before ‘_U_’
/home/devel/dev/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c: In function ‘proto_reg_handoff_btbb’:
/home/devel/dev/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c:573:44: error: ‘dissect_btbb’ undeclared (first use in this function)
/home/devel/dev/libbtbb-2015-10-R1/wireshark/plugins/btbb/packet-btbb.c:573:44: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [CMakeFiles/btbb.dir/packet-btbb.c.o] Error 1
make[1]: *** [CMakeFiles/btbb.dir/all] Error 2
make: *** [all] Error 2
```

Same applies to the btbredr plugin.

```
Package: wireshark
Status: install ok installed
Priority: optional
Section: net
Installed-Size: 2270
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: amd64
Version: 1.6.7-1
Replaces: ethereal (<< 1.0.0-3)
Depends: libc6 (>= 2.15), libgdk-pixbuf2.0-0 (>= 2.22.0), libglib2.0-0 (>= 2.24.0), libgtk2.0-0 (>= 2.18.0), libpango1.0-0 (>= 1.14.0), libpcap0.8 (>= 0.9.8), libportaudio2 (>= 19+svn20101113), libwireshark1 (>= 1.6.0-1), libwiretap1 (>= 1.6.0-1), libwsutil1 (>= 1.6.0-1), zlib1g (>= 1:1.1.4), wireshark-common (= 1.6.7-1)
```

```
Package: libpcap-dev
Status: install ok installed
Priority: optional
Section: libdevel
Installed-Size: 35
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: all
Source: libpcap
Version: 1.1.1-10
Depends: libpcap0.8-dev
```

Please help!

dominicgs, 2015-12-09:
I think the problem is that Wireshark 1.6 is just too old for us to support.  Which version of ubuntu is this packaged with?

If you really need to build the dissectors for that version, we may be able to help you work around the problem, but I think we'll find that we're using functions/#defines that don't exist in the older Wireshark header files.

maximcherny, 2015-12-09:
Thanks, I managed to build successfully on Kali 2.0 Light using the distro wireshark and wireshark-dev.

Closed #33.

closed 2016-01-21 by dominicgs