greatscottgadgets/libbtbb #36: Kismet pcapbtbb file only contains 14 byte packets

jakupKdG, 2016-02-20:
Hi,

I believe i have set everything up the way i should with kismet, ubertooth, libbtbb and wireshark.
I followed tutorials and installed kismet-2013-03-R1b, ubertooth-2015-10-R1 and libbtbb-2015-10-R1.

However when i read the pcapbtbb output file with wireshark, it only contains 14 byte longs packets with no destination. I read in previous issues that the lack of destination address is not unnusual because the packets don't always contain it. But i can't seem to find anything that helps me with the issue of not finding any information in the packets or that they are 14 bytes long.

Here is a picture of the file opened in wireshark.
(https://cloud.githubusercontent.com/assets/17140356/13199028/79d8f548-d819-11e5-88d1-ff6fffb70ed7.png)

I tried to send text between 2 devices and even tried to send music to a bluetooth headset, with the thought that a text message would be send to quickly and the ubertooth wouldn't have enough time to sniff it. But that didn't show any different result
 
Does anyone know how to help me with this?

dominicgs, 2016-02-21:
The packets at the start of your capture all appear to be ID packets, these are always 14 bytes in length and use the GIAC address - 9E:8B:33.  This is expected for devices performing discovery.

What do the packets later in the capture look like?

jakupKdG, 2016-02-21:
All packets look the same, the only difference between them is the source address which varies between ...2A:EC and ...9E:8B:33.

dominicgs, 2016-02-21:
9E:8B:33 is the GIAC, so you can ignore all of those packets for now.  The others only have 2 bytes of address?  I would expect a minimum of 3 and, if we're lucky 4.

jakupKdG, 2016-02-21:
No srr i left the other parts out because they are all zeros. 00:00:00_00:2A:EC is the complete source address.

dominicgs, 2016-02-21:
The 00:2A:EC is a valid LAP, but you're probably missing the UAP.  You may have more luck with `ubertooth-rx -l 002aec`

mateiclaudiu, 2016-02-21:
I have the same problem as @jakupKdG . Is there a way to save the packets in a file with `ubertooth-rx`?

dominicgs, 2016-02-21:
@mateiclaudiu You can use -r or -q for pcap(ng) files, but the code is definitely buggy.

mateiclaudiu, 2016-02-21:
I am actually sending a string message between two devices for ex. "Hallo World!" and I want to sniff and get the packets. Is there a way to read the sended message with ubertooth-rx ?

dominicgs, 2016-02-22:
ubertooth-rx, if given a LAP, will attempt to find the UAP and CLK then follow the connection.  This is certainly not the most reliable feature of Ubertooth, so I can't make any guarantees.

jakupKdG, 2016-02-22:
If i have the full BL_ADDR of the device (LAP and UAP), is it possible to for example get that "hallo world" @mateiclaudiu is reffering to that could be sent between devices?

dominicgs, 2016-02-22:
There are multiple stages to the Bluetooth sniffing process.  The following steps produce the given results:
* Sniff promiscuously (`ubertooth-rx`) -> LAPs
* Sniff multiple packets with the same LAP (`ubertooth-rx -l <LAP>`) -> UAP
* That stage will then automatically go on to deriving CLK -> CLK

At this point we can attempt to follow the hopping pattern and hop along with the piconet.  From here on you will get as much data as possible, but there are a lot of things that affect the data received.  These include (but areprobably not limited to):
* Encryption
* Clock skew (git code has some features for compensating for this)
* multi-slot packet types
* EDR packets
* Adaptive Frequency Hopping (AFH)

If you're repeatedly sending the data between the devices, using short packet types and have no encryption or AFH enabled, then you stand a relatively good chance of seeing the data.

hectorbm94, 2016-05-08:
Hello @dominicgs is there any way of saving the result of the command "ubertooth-rx -l <LAP>" so that I can see it on Wireshark?

dominicgs, 2016-05-08:
@hectorbm94 No, the data from `ubertooth-rx -l` does not contain full packets and any that are present are whitened, so cannot be decoded.

straithe, 2021-02-17:
I'm going to close this issue as there hasn't been a response in a while, but please re-open this issue or open a new one if you still need assistance.

closed 2021-02-17 by straithe