greatscottgadgets/libbtbb #37: Usage error of ubertooth-rx and save to pcap file

michaelp123, 2016-02-22:
When I try ubertooth-rx -q mm.pcap

I get the following error:

PCAP error:: mm.pcap: link-layer type 255 isn't supported in savefiles
ubertooth-rx: btbb_pcap_create_file: : Success

I run the latest libpcap and libbtbb releases and still get this error.

Would love to get some help.

dominicgs, 2016-02-22:
Which version of libpcap, libbtbb and libubertooth are you using?

Pcap support is fairly new and is rather experimental at this stage.  The git code has better support than the last release, but it is not currently stable.

michaelp123, 2016-02-22:
Hi I am using libpcap 1.7.4.
The libbtbb I downloaded from here: https://github.com/greatscottgadgets/libbtbb.
And also using ubertooth 2015-10-R1

dominicgs, 2016-02-22:
Did you download a release of libbtbb or clone the git repository?

We recommend using matched release versions for libbtbb and Ubertooth, or using git latest for both (as long as they match).

michaelp123, 2016-02-22:
I downloaded the Zip file. It is called libbtbb-master.
Is it correct?

michaelp123, 2016-02-22:
Can please tell me what is proper way to download both packages?

dominicgs, 2016-02-22:
That may cause your error.  You can download the 2015-10-R1 release from here: https://github.com/greatscottgadgets/libbtbb/releases

michaelp123, 2016-02-22:
Hi, thanks.
Installed this version. Still getting this error.

dominicgs, 2016-02-22:
You may have multiple versions installed - `ls -l /usr/local/lib/libbtbb*` should show a few files. It's probably best to delete them all and then re-run `make install`.

Have you tried using pcapng files rather than pcap? `ubertooth-rx -r mm.pcap`

dominicgs, 2016-02-22:
This code is certainly buggy and not stable yet, but I have not seen your original error before, so I am trying to determine if it's a new bug or if it is something caused by the environment.

michaelp123, 2016-02-22:
Hi, in /usr/local/lib/ I see three files: libbtbb.so, libbtbb.so.0, libbtbb.so.0.3. (I deleted them and re-ran make install). After that these 3 files appeared again.

When trying pcapng, I get no error, the file is saved but in wireshark I get: "22 WTAP_ENCAP=0"

michaelp123, 2016-02-22:
Do I need to install something else?

dominicgs, 2016-02-22:
Have you installed the Wireshark dissectors from the
libbtbb/wireshark/plugins/ directory?

Again, I am not confident that this code works reliably.

michaelp123, 2016-02-22:
Yes I installed all dissectors and still getting "WTAP_ENCAP=0".

Is there any other way I can see payload from bluetooh packets?

dominicgs, 2016-02-22:
`ubertooth-rx` or `ubertooth-follow` will log data to screen if you specify the LAP/UAP on the commandline.  As always, the following is not completely reliable, so it is unlikely that you will get all data.

Any encrypted packets will still be encrypted.

che88, 2016-03-14:
i have also this error running this command ubertooth-rx -q trace.pcap.
link-layer type 255 isn't supported in savefiles

is this bug fixed now?
i've installed my ubertooth with the ubertooth-2015-10-R1 and libbtbb-2015-10-R1 release

dominicgs, 2016-03-29:
No, this bug will be closed when it is fixed.

There is a chance that it will not occur with the version of libbtbb and Ubertooth in the git repositories, but I have not confirmed it.