greatscottgadgets/libbtbb #39: BTLE Time Stamp Issue

harryh3o, 2016-02-27:
I'm seeing an issue with time stamps when doing BTLE captures on the Ubertooth One.
I see differences within just a few minutes of the capture running, and when I let it run for days, it gains days in the time stamp. For example I started a capture on 2/24/16 22:30:36 and stopped it at 2/27/2016 18:50:32.  The last packet captured is time stamped 2/29/16 03:12:11. So roughly 33+ hours ahead on time.
System details:
RaspberryPi2
Kali Linux (kali-2.1-rpi2)
Ubertooth One, Firmware Revision 2015-10-R1

Running the capture using the following commands, both displaying the same timestamp problem,
ubertooth-btle -f -c filename.pcpap
ubertooth-btle -f -q filename.pcpap

anotherfiz, 2017-01-12:
I'm seeing this issue as well with BTLE captures on the Ubertooth One.  The only difference between the above poster's experience and my own is that I am running captures for a much smaller amount of time.  In a 250 second capture, I see anomalous timestamps ~530 seconds into the future.  For example: Packet 848 has a timestamp of 8:40am, and packet 849 has a timestamp of 8:50am even though the captures stopped at 8:45am

System details:
Dell Precision Tower 5810
Ubuntu 16.04 LTS
Ubertooth One, Firmware revision 2015-10-R1

smbeyer8, 2017-08-08:
I am also seeing this issue on the latest firmware (2017-03-R2 (API:1.02)). I see timing drift on one ubertooth and timing issues among uberteeth when starting three at the same time (i.e. each ubertooth listening on a different advertisement channel starting at the same time).

straithe, 2021-05-04:
@harryh3o Are you still experiencing this issue?

straithe, 2021-06-08:
I'm going to close this as there hasn't been a response in the last month, but please re-open this issue or open a new one if you still need assistance.

closed 2021-06-08 by straithe