greatscottgadgets/libbtbb #40: issue with ubertooth-btle

michaelp123, 2016-02-28:
Hi I am using "ubertooth-btle -p -c capture.pcap" command and getting this error:

"control message unsupported"

I am using ubertooth 2015-10-R1 version.

What could be the problem?

dominicgs, 2016-02-28:
What does ubertooth-util -vV show?

Your firmware may need updating.

michaelp123, 2016-02-28:
It shows I am using 2015-10-R1 version.

dominicgs, 2016-02-28:
For both host and firmware? I'm not sure how we'd get an unsupported
command with matched versions.

Could you run the command again but put $LIBUSB_DEBUG=8  in front of it?
This should tell us which command is failing.

michaelp123, 2016-02-28:
running ubertooth-util -v shows: "Firmware revision: 2015-10-R1"
running ubertooth-util -V shows: "ubertooth 2015-10-R1 (dominicgs@hydrogen) Mon Oct 12 22:30:57 BST 2015"

I also ran LIBUSB_DEBUG=8 in front of the command but I see many text.

dominicgs, 2016-02-28:
I'm interested in the 10 or so lines before the "control message unsupported" error.  Could you copy them in to this issue?

michaelp123, 2016-02-28:
libusbx: debug [usbi_handle_transfer_completion] transfer 0x1cc7c28 has callback 0x7f5747c857f0
libusbx: debug [sync_transfer_cb] actual_length=1
libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
libusbx: debug [handle_events] poll() returned 1
libusbx: debug [reap_for_handle] urb type=2 status=0 transferred=1
libusbx: debug [handle_control_completion] handling completion status 0
libusbx: debug [disarm_timerfd] 
libusbx: debug [usbi_handle_transfer_completion] transfer 0x1cc7c28 has callback 0x7f5747c857f0
libusbx: debug [sync_transfer_cb] actual_length=1
libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
libusbx: debug [handle_events] poll() returned 1
libusbx: debug [reap_for_handle] urb type=2 status=0 transferred=1
libusbx: debug [handle_control_completion] handling completion status 0
libusbx: debug [disarm_timerfd] 
libusbx: debug [usbi_handle_transfer_completion] transfer 0x1cc7c28 has callback 0x7f5747c857f0
libusbx: debug [sync_transfer_cb] actual_length=1
libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
libusbx: debug [handle_events] poll() returned 1
libusbx: debug [reap_for_handle] urb type=2 status=0 transferred=1
libusbx: debug [handle_control_completion] handling completion status 0
libusbx: debug [disarm_timerfd] 
libusbx: debug [usbi_handle_transfer_completion] transfer 0x1cc7c28 has callback 0x7f5747c857f0
libusbx: debug [sync_transfer_cb] actual_length=1
libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
libusbx: debug [handle_events] poll() returned 1
libusbx: debug [reap_for_handle] urb type=2 status=0 transferred=1
libusbx: debug [handle_control_completion] handling completion status 0
libusbx: debug [disarm_timerfd] 
libusbx: debug [usbi_handle_transfer_completion] transfer 0x1cc7c28 has callback 0x7f5747c857f0
libusbx: debug [sync_transfer_cb] actual_length=1
libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
libusbx: debug [handle_events] poll() returned 1
libusbx: debug [reap_for_handle] urb type=2 status=0 transferred=1
libusbx: debug [handle_control_completion] handling completion status 0
libusbx: debug [disarm_timerfd] 
libusbx: debug [usbi_handle_transfer_completion] transfer 0x1cc7c28 has callback 0x7f5747c857f0
libusbx: debug [sync_transfer_cb] actual_length=1
libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
libusbx: debug [handle_events] poll() returned 1
libusbx: debug [reap_for_handle] urb type=2 status=0 transferred=1
libusbx: debug [handle_control_completion] handling completion status 0
libusbx: debug [disarm_timerfd] 
libusbx: debug [usbi_handle_transfer_completion] transfer 0x1cc7c28 has callback 0x7f5747c857f0
libusbx: debug [sync_transfer_cb] actual_length=1
libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
libusbx: debug [handle_events] poll() returned 1
libusbx: debug [reap_for_handle] urb type=2 status=0 transferred=1
libusbx: debug [handle_control_completion] handling completion status 0
libusbx: debug [disarm_timerfd] 
libusbx: debug [usbi_handle_transfer_completion] transfer 0x1cc7c28 has callback 0x7f5747c857f0
libusbx: debug [sync_transfer_cb] actual_length=1
libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
libusbx: debug [handle_events] poll() returned 1
libusbx: debug [reap_for_handle] urb type=2 status=0 transferred=1
[... 287 lines cut ...]
libusbx: debug [reap_for_handle] urb type=2 status=0 transferred=1
libusbx: debug [handle_control_completion] handling completion status 0
libusbx: debug [disarm_timerfd] 
libusbx: debug [usbi_handle_transfer_completion] transfer 0x1cc7c28 has callback 0x7f5747c857f0
libusbx: debug [sync_transfer_cb] actual_length=1
libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
libusbx: debug [handle_events] poll() returned 1
libusbx: debug [reap_for_handle] urb type=2 status=0 transferred=1
[... cut ...]
libusbx: debug [reap_for_handle] urb type=2 status=0 transferred=1
libusbx: debug [handle_control_completion] handling completion status 0
libusbx: debug [disarm_timerfd] 
libusbx: debug [usbi_handle_transfer_completion] transfer 0x1cc7c28 has callback 0x7f5747c857f0libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
^Clibusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
libusbx: debug [handle_events] poll() returned 1

michaelp123, 2016-02-28:
I ran it like this: LIBUSB_DEBUG=8 ubertooth-btle -p -c capture.pcap

dominicgs, 2016-02-28:
Unfortunately that output doesn't contain the "control message unsupported" error.  In order to debug the issue, I need to know which control message the host is sending that the firmware does not support.

michaelp123, 2016-02-28:
So what else can I do in order to debug?

dominicgs, 2016-02-28:
This is the information that I need to track down this bug.  I am not able to reproduce this issue, so I need to know which command is failing.

I would suggest running the command until you get the error message again.  Probably like this:
`LIBUSB_DEBUG=8 ubertooth-btle -p -c capture.pcap 2>&1 | tee ble_usb.log | grep "control message unsupported"`

michaelp123, 2016-02-29:
I ran it this way and the only thing I get is "control message unsupported"

dominicgs, 2016-02-29:
The command should have produced a log file called `ble_usb.log`.  That file should include the debug information that I need.

michaelp123, 2016-02-29:
[ble_log.txt](https://github.com/greatscottgadgets/libbtbb/files/150606/ble_log.txt)

michaelp123, 2016-02-29:
Attached the log file.

dominicgs, 2016-02-29:
The relevant part seems to be here:
```
libusbx: debug [linux_get_device_address] getting address for device: usb4 detached: 0
libusbx: debug [linux_get_device_address] scan usb4
libusbx: debug [linux_get_device_address] bus=4 dev=1
libusbx: debug [linux_enumerate_device] busnum 4 devaddr 1 session_id 1025
libusbx: debug [linux_enumerate_device] allocating new device for 4/1 (session 1025)
libusbx: debug [linux_get_device_address] getting address for device: 4-1 detached: 0
libusbx: debug [linux_get_device_address] scan 4-1
libusbx: debug [linux_get_device_address] bus=4 dev=2
libusbx: debug [linux_enumerate_device] busnum 4 devaddr 2 session_id 1026
libusbx: debug [linux_enumerate_device] allocating new device for 4/2 (session 1026)
libusbx: debug [linux_get_parent_info] Dev 0x9877e0 (4-1) has parent 0x987310 (usb4) port 1
libusbx: debug [usbi_add_pollfd] add fd 7 events 1
libusbx: debug [usbi_add_pollfd] add fd 9 events 1
libusbx: debug [usbi_io_init] using timerfd for timeouts
libusbx: debug [usbi_add_pollfd] add fd 11 events 1
libusbx: debug [libusb_get_device_list] 
libusbx: debug [discovered_devs_append] need to increase capacity
libusbx: debug [libusb_get_device_descriptor] 
libusbx: debug [libusb_get_device_descriptor] 
libusbx: debug [libusb_get_device_descriptor] 
libusbx: debug [libusb_get_device_descriptor] 
libusbx: debug [libusb_get_device_descriptor] 
libusbx: debug [libusb_get_device_descriptor] 
libusbx: debug [libusb_get_device_descriptor] 
libusbx: debug [libusb_get_device_descriptor] 
libusbx: debug [libusb_get_device_descriptor] 
libusbx: debug [libusb_get_device_descriptor] 
libusbx: debug [libusb_get_device_descriptor] 
libusbx: debug [libusb_get_device_descriptor] 
libusbx: debug [libusb_get_device_descriptor] 
libusbx: debug [libusb_get_device_descriptor] 
libusbx: debug [libusb_open] open 1.2
libusbx: debug [usbi_add_pollfd] add fd 12 events 4
libusbx: debug [libusb_claim_interface] interface 0
libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
libusbx: debug [handle_events] poll() returned 1
libusbx: debug [reap_for_handle] urb type=2 status=-32 transferred=0
libusbx: debug [handle_control_completion] handling completion status -32
libusbx: debug [handle_control_completion] unsupported control request
libusbx: debug [disarm_timerfd] 
libusbx: debug [usbi_handle_transfer_completion] transfer 0x981bb8 has callback 0x7f7692fc07f0
libusbx: debug [sync_transfer_cb] actual_length=0
control message unsupported
```

@mikeryan This is ubertooth-btle, do you have any ideas?

straithe, 2021-02-17:
Are you still experiencing this issue?

closed 2021-03-24 by straithe