greatscottgadgets/libbtbb #41: [libbtbb-2015-10-R1]  Wireshark plugin (now legacy-plugin) build failing.

djwbrown, 2016-05-24:
In `packet-btbb.c` I needed to comment out line 259:
`/* pinfo->private_data = &fake_acl_data; */`

This allowed the plugin to build, but I have no idea what I may have broken in the process. It's preceded by a 'FIXME' and the `struct packet_info` has no member `private_data`.

dominicgs, 2016-05-24:
Which version of Wireshark are you building against?  This looks like the build error that caused the legacy/non-legacy split to be added.

You can grab the newer plugins from git without needing to rebuild libbtbb.

I plan to release this in the next two weeks.

djwbrown, 2016-05-25:
OK thanks! I've got wireshark-2.0.3, and I see that the line has been changed in the newer (non-legacy) version of `packet-btbb.c` on master.

djwbrown, 2016-05-25:
I also had to manually remove the `btbb.so` `btbredr.so` plugins from /usr/lib/x86_64-linux-gnu/wireshark/libwireshark3/plugins/

closed 2016-05-25 by djwbrown