greatscottgadgets/libbtbb #46: wireshark plugins install

topeminent, 2016-06-19:
Hi,

I m trying to install on Kali 2016-01 the wireshark btbb and br/edr plugins, without success. I know that the directory passed to cmake differs and on mine its in the /usr/lib/x86_64-linux-gnu/wireshark/plugins/2.0.4

however if i run cmake -DCMAKE_INSTALL_LIBDIR=/usr/lib/x86_64-linux-gnu/wireshark/plugins/2.0.4 ..

this is the output:

root@kali:~/libbtbb-2015-10-R1/wireshark/plugins/btbredr/build# make
[ 20%] Generating plugin.c
Scanning dependencies of target btbredr
[ 40%] Building C object CMakeFiles/btbredr.dir/plugin.c.o
[ 60%] Building C object CMakeFiles/btbredr.dir/packet-btbredr.c.o
In file included from /usr/include/wireshark/epan/packet.h:26:0,
                 from /root/libbtbb-2015-10-R1/wireshark/plugins/btbredr/packet-btbredr.c:31:
/root/libbtbb-2015-10-R1/wireshark/plugins/btbredr/packet-btbredr.c: In function ‘dissect_payload_header1’:
/root/libbtbb-2015-10-R1/wireshark/plugins/btbredr/packet-btbredr.c:179:19: warning: implicit declaration of function ‘tvb_length_remaining’ [-Wimplicit-function-declaration]
  DISSECTOR_ASSERT(tvb_length_remaining(tvb, offset) >= 1);
                   ^
/usr/include/wireshark/epan/proto.h:165:13: note: in definition of macro ‘DISSECTOR_ASSERT’
   ((void) ((expression) ? (void)0 : \
             ^
/root/libbtbb-2015-10-R1/wireshark/plugins/btbredr/packet-btbredr.c: In function ‘dissect_dm1’:
/root/libbtbb-2015-10-R1/wireshark/plugins/btbredr/packet-btbredr.c:289:9: error: ‘packet_info {aka struct _packet_info}’ has no member named ‘private_data’
    pinfo->private_data = &fake_acl_data;
         ^
/root/libbtbb-2015-10-R1/wireshark/plugins/btbredr/packet-btbredr.c: In function ‘dissect_btbredr’:
/root/libbtbb-2015-10-R1/wireshark/plugins/btbredr/packet-btbredr.c:317:6: warning: implicit declaration of function ‘tvb_length’ [-Wimplicit-function-declaration]
  if (tvb_length(tvb) > 0 && tvb_length(tvb) < 9)
      ^
CMakeFiles/btbredr.dir/build.make:93: recipe for target 'CMakeFiles/btbredr.dir/packet-btbredr.c.o' failed
make[2]: *** [CMakeFiles/btbredr.dir/packet-btbredr.c.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/btbredr.dir/all' failed
make[1]: *** [CMakeFiles/btbredr.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Any guidance would be appreciated.

dominicgs, 2016-06-20:
This is a change for Wireshark 2.0+.  There is a fix in git which does not require building libbtbb from git (don't build libbtbb from git, it will cause compatibility problems).

You will find the updated plugins here: https://github.com/greatscottgadgets/libbtbb/tree/master/wireshark/plugins

topeminent, 2016-06-20:
Thank you for your answer Dominic. I was able to install the plugin using the right one for wireshark 2.0.4. 
Still i have another question, regarding the installation of kismet with ubertooth plugin. i was able to install it, however the plugins are not loaded. Root user is member of kismet group, and i enabled plugins in the config file. Same thing happened at the installation of version 2016-01-r1, as well as installing the 2013-03-r1b version, as illustrated in the picture 
[image]

topeminent, 2016-06-20:
Thank you for your answer Dominic. I was able to install the plugin using the right one for wireshark 2.0.4. First time i m using it.

topeminent, 2016-06-20:
However moving one step further. (i am really noob regarding this) i installed kismet v. 2013-03. and it seem like the plugins are not loading in. I added my user to kismet group, and i enabled pcapbtbb in config file. 
[image]
And my third question would be, in kismet which interface should i define to listen (if i type ubertooth, it s not recognized)

Thank you for your help,
Steve

dominicgs, 2016-06-21:
Could you try running `kismet_server -c ubertooth` and copying the output in to this issue?

topeminent, 2016-06-21:
root@kali:~# kismet_server -c ubertooth
ERROR: Kismet was started as root, NOT launching external control binary.  
       This is NOT the preferred method of starting Kismet as Kismet will 
       continue to run as root the entire time.  Please read the README 
       file section about Installation & Security and be sure this is what 
       you want to do.
INFO: Reading from config file /usr/local/etc/kismet.conf
INFO: No 'dronelisten' config line and no command line drone-listen 
      argument given, Kismet drone server will not be enabled.
INFO: Created alert tracker...
INFO: Creating device tracker...
INFO: Registered 80211 PHY as id 0
ERROR: Failed to open user plugin directory (/root/.kismet//plugins/): No 
       such file or directory
INFO: Loaded info for plugin 'alertsyslog.so': Plugin name: 'ALERTSYSLOG' 
      Plugin version: '2013-03-R0' Plugin description: 'ALERTSYSLOG Plugin'
INFO: Activated plugin '/usr/local/lib/kismet/alertsyslog.so': 
      'ALERTSYSLOG' '2013-03-R0' 
INFO: Loaded info for plugin 'btscan.so': Plugin name: 'BTSCAN' Plugin 
      version: '2013-03-R0' Plugin description: 'Active Bluetooth scanning 
      plugin'
INFO: Activated plugin '/usr/local/lib/kismet/btscan.so': 'BTSCAN' 
      '2013-03-R0' 
INFO: Loaded info for plugin 'kismet-syslog.so': Plugin name: 'SYSLOG' 
      Plugin version: '2013-03-R0' Plugin description: 'SYSLOG Plugin'
ERROR: Kismet-Syslog not redirecting any messages to syslog because no 
       syslogtype= directive found in kismet.conf
INFO: Kismet-Syslog redirecting no messages to syslog
INFO: Activated plugin '/usr/local/lib/kismet/kismet-syslog.so': 'SYSLOG' 
      '2013-03-R0' 
INFO: Loaded info for plugin 'ubertooth.so': Plugin name: 'UBERTOOTH' 
      Plugin version: '2013-03-R0' Plugin description: 'Ubertooth plugin 
      for Bluetooth baseband protocol'
INFO: Pcap logging for type pcapbtbb
INFO: Opened pcapdump log file 'Kismet-20160621-23-42-51-1.pcapbtbb'
INFO: Activated plugin '/usr/local/lib/kismet/ubertooth.so': 'UBERTOOTH' 
      '2013-03-R0' 
INFO: Loaded info for plugin 'spectool_net.so': Plugin name: 'SPECTOOL' 
      Plugin version: '2013-03-R0' Plugin description: 'Spectool-Net'
INFO: Kismet will spend extra time on channels 1,6,11
INFO: Kismet will attempt to hop channels at 3 channels per second unless 
      overridden by source-specific options
INFO: Matched source type 'ubertooth' for auto-type source 'ubertooth'
INFO: Using default channel list 'UBERTOOTH' on source 'ubertooth'
INFO: Ubertooth using first USB device that looks like a Ubertooth
INFO: Created source ubertooth with UUID 
      6522bb6c-382b-11e6-99cb-b8072e49dc03
INFO: Disabling channel hopping on source 'ubertooth' because it is not 
      capable of setting the channel.
INFO: Will attempt to reopen on source 'ubertooth' if there are errors
INFO: Created TCP listener on port 2501
INFO: Kismet drone framework disabled, drone will not be activated.
INFO: Inserting basic packet dissectors...
INFO: Allowing Kismet frontends to view WEP keys
INFO: Starting GPS components...
INFO: Enabling reconnection to the GPS device if the link is lost
INFO: Using GPSD server on localhost:2947
ERROR: Could not open OUI file '/etc/manuf': No such file or directory
ERROR: Could not open OUI file '/usr/share/wireshark/wireshark/manuf': No 
       such file or directory
INFO: Opened OUI file '/usr/share/wireshark/manuf
INFO: Indexing manufacturer db
INFO: Completed indexing manufacturer db, 28250 lines 565 indexes
INFO: Creating network tracker...
INFO: Creating channel tracker...
INFO: Registering dumpfiles...
INFO: Pcap log in PPI format
INFO: Opened pcapdump log file 'Kismet-20160621-23-42-51-1.pcapdump'
INFO: Opened netxml log file 'Kismet-20160621-23-42-51-1.netxml'
INFO: Opened nettxt log file 'Kismet-20160621-23-42-51-1.nettxt'
INFO: Opened gpsxml log file 'Kismet-20160621-23-42-51-1.gpsxml'
INFO: Opened alert log file 'Kismet-20160621-23-42-51-1.alert'
INFO: No spectools= line in config file, will not try to use spectools for 
      spectrum data
INFO: Activated plugin '/usr/local/lib/kismet/spectool_net.so': 'SPECTOOL' 
      '2013-03-R0' 
INFO: Kismet starting to gather packets
INFO: Started source 'ubertooth'
rx blocks of 64 bytes in 512 byte transfers

Question nr 3 is sorted out, i installed incorrectly, with an incorrect symbolic link .
However still cant load client plugins

SagiSatish, 2016-08-06:
@topeminent I am having the same error. Got any solution ?

dominicgs, 2016-08-11:
It may well be a library version of something that has changed with the latest Kali.  I'm not sure where to start looking though.

topeminent, 2016-09-05:
Yes, it s probably kali s fault. Got it to work on ubuntu 16.04
[image] Thanks for the tip

closed 2016-09-06 by topeminent