greatscottgadgets/libbtbb #50: Wireshark plugins btbb and btbredr not working/compiling with latest Wireshark 2.2

xk4l23, 2016-09-15:
When compiling the Wireshark plugings for WS 2.2, I get the follwing compile warnings:
For btbb ...

```
[ 20%] Generating plugin.c
Scanning dependencies of target btbb
[ 40%] Building C object CMakeFiles/btbb.dir/plugin.c.o
[ 60%] Building C object CMakeFiles/btbb.dir/packet-btbb.c.o
/home/b/Temp/ubertooth/libbtbb/wireshark/plugins/btbb/packet-btbb.c: In function ‘proto_reg_handoff_btbb’:
/home/b/Temp/ubertooth/libbtbb/wireshark/plugins/btbb/packet-btbb.c:572:16: warning: implicit declaration of function ‘new_create_dissector_handle’ [-Wimplicit-function-declaration]
  btbb_handle = new_create_dissector_handle(dissect_btbb, proto_btbb);
                ^
/home/b/Temp/ubertooth/libbtbb/wireshark/plugins/btbb/packet-btbb.c:572:14: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
  btbb_handle = new_create_dissector_handle(dissect_btbb, proto_btbb);
              ^
[ 80%] Building C object CMakeFiles/btbb.dir/packet-btbrlmp.c.o
/home/b/Temp/ubertooth/libbtbb/wireshark/plugins/btbb/packet-btbrlmp.c: In function ‘proto_register_btbrlmp’:
/home/b/Temp/ubertooth/libbtbb/wireshark/plugins/btbb/packet-btbrlmp.c:2864:32: warning: passing argument 2 of ‘register_dissector’ from incompatible pointer type [-Wincompatible-pointer-types]
  register_dissector("btbrlmp", dissect_btbrlmp, proto_btbrlmp);
                                ^
In file included from /home/b/Temp/ubertooth/libbtbb/wireshark/plugins/btbb/packet-btbrlmp.c:31:0:
/usr/include/wireshark/epan/packet.h:490:34: note: expected ‘dissector_t {aka int (*)(struct tvbuff *, struct _packet_info *, struct _proto_node *, void *)}’ but argument is of type ‘void (*)(tvbuff_t *, packet_info *, proto_tree *) {aka void (*)(struct tvbuff *, struct _packet_info *, struct _proto_node *)}’
 WS_DLL_PUBLIC dissector_handle_t register_dissector(const char *name, dissector
                                  ^
[100%] Linking C shared module btbb.so
[100%] Built target btbb
```

And for btbredr ...

```
[ 20%] Generating plugin.c
Scanning dependencies of target btbredr
[ 40%] Building C object CMakeFiles/btbredr.dir/plugin.c.o
[ 60%] Building C object CMakeFiles/btbredr.dir/packet-btbredr.c.o
/home/b/Temp/ubertooth/libbtbb/wireshark/plugins/btbredr/packet-btbredr.c: In function ‘proto_reg_handoff_btbredr’:
/home/b/Temp/ubertooth/libbtbb/wireshark/plugins/btbredr/packet-btbredr.c:651:19: warning: implicit declaration of function ‘new_create_dissector_handle’ [-Wimplicit-function-declaration]
  btbredr_handle = new_create_dissector_handle(dissect_btbredr,
                   ^
/home/b/Temp/ubertooth/libbtbb/wireshark/plugins/btbredr/packet-btbredr.c:651:17: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
  btbredr_handle = new_create_dissector_handle(dissect_btbredr,
                 ^
[ 80%] Building C object CMakeFiles/btbredr.dir/packet-btlmp.c.o
/home/b/Temp/ubertooth/libbtbb/wireshark/plugins/btbredr/packet-btlmp.c: In function ‘proto_register_btlmp’:
/home/b/Temp/ubertooth/libbtbb/wireshark/plugins/btbredr/packet-btlmp.c:2864:30: warning: passing argument 2 of ‘register_dissector’ from incompatible pointer type [-Wincompatible-pointer-types]
  register_dissector("btlmp", dissect_btlmp, proto_btlmp);
                              ^
In file included from /home/b/Temp/ubertooth/libbtbb/wireshark/plugins/btbredr/packet-btlmp.c:31:0:
/usr/include/wireshark/epan/packet.h:490:34: note: expected ‘dissector_t {aka int (*)(struct tvbuff *, struct _packet_info *, struct _proto_node *, void *)}’ but argument is of type ‘void (*)(tvbuff_t *, packet_info *, proto_tree *) {aka void (*)(struct tvbuff *, struct _packet_info *, struct _proto_node *)}’
 WS_DLL_PUBLIC dissector_handle_t register_dissector(const char *name, dissector
                                  ^
[100%] Linking C shared module btbredr.so
[100%] Built target btbredr
```

Wiresharks compaints when started that the plugins cannot be started ...

```
Couldn't load module /usr/lib/i386-linux-gnu/wireshark/plugins/2.2.0/btbredr.so: /usr/lib/i386-linux-gnu/wireshark/plugins/2.2.0/btbredr.so: undefined symbol: new_create_dissector_handle

Couldn't load module /usr/lib/i386-linux-gnu/wireshark/plugins/2.2.0/btbb.so: /usr/lib/i386-linux-gnu/wireshark/plugins/2.2.0/btbb.so: undefined symbol: new_create_dissector_handle
```

dominicgs, 2016-09-18:
This looks like the Wireshark dissector API has changed again.  I'll investigate the changes and try to create something that builds against the latest Wireshark.

The btbredr plugin should be available in Wireshark releases very soon (which will solve this problem for good)

xk4l23, 2016-09-19:
Sounds great, many thanks, Dom :-)

chrdub, 2016-10-10:
I'm having the same issue. Should I revert back to an older version of Wireshark, and if so, which version?

dominicgs, 2016-10-11:
I don't believe that you should need to build the plugins for the latest version of Wireshark.

chrdub, 2016-10-11:
I have an older version of btbb.so that I had built for Wireshark 1.11 or 1.12, but if I try to use it with Wireshark 2.2.1, Wireshark crashes with the following message:
"Err   Field 'Meta Data' (bbtb.meta) is an FT_NONE but has a bitmask"

dominicgs, 2016-10-11:
You do not need to build the plugins for Wireshark 2.2+

chrdub, 2016-10-11:
Are the plugins part of the release? I don't see them in the list of plugins when I run Wireshark. It processes the pcapbtbb captures as "Ethernet II".

dominicgs, 2016-10-19:
That's an older file format.  You would need to build the legacy plugins.  We do not plan to maintain the legacy plugins for Wireshark 2.2+

chrdub, 2016-10-19:
How do I get to the newer file format? I've compiled the Kismet plugin
using the latest libbtbb.

muzzymurray, 2016-12-09:
Sorry I am just trying to understand where this bug is at. When I open btbb-sample-mouse.pcap, the traffic in wireshark is only showing Ethernet ||. When you say that there are no plans to maintain the plugins for 2.2+ this means that we should try to work with an older wireshark release.

dominicgs, 2017-03-04:
The plugins required vary based on the Wireshark version:
Build [legacy](https://github.com/greatscottgadgets/libbtbb/tree/master/wireshark/plugins-legacy) for pre-2.0

Build the [regular plugins](https://github.com/greatscottgadgets/libbtbb/tree/master/wireshark/plugins) for Wireshark 2.0-2.2

You should not need to build any plugins for Wireshark 2.2+

closed 2017-03-04 by dominicgs

Silberwoelfin, 2017-04-19:
Using Wireshark 2.2.5 the above issue is still present. The plugins won't work, and without them Wireshark does not recognize the protocol (0xfff0) and only shows the frames as "Ethernet II".

dominicgs, 2017-04-21:
The newer file format should be supported directly by Wireshark 2.2.5.  The version that shows up at "Ethernet II" is the older format, it has been retired.

jgalmeidah, 2017-05-02:
Just to be clear, the btbb-sample-mouse.pcap file found in libbtbb/wireshark/plugins is "in the older file format" so Wireshark 2.2+ will still decode it as Ethernet II.  If capturing new data using Kismet and opening that in Wireshark 2.2+ will decode properly ?

jgalmeidah, 2017-05-02:
Using libbtbb 2017-03-R2; Kismet 2016-07-R1 and Wireshark 2.2.6

dominicgs, 2017-05-02:
I believe that should work, however the kismet plugins are fairly out of date with respect to kismet development at this point.  I can't be certain that they will work or output the newer file format.

jgalmeidah, 2017-05-02:
So what do you suggest to get BT classic data on Wireshark (any version)?

dominicgs, 2017-05-02:
`ubertooth-rx -l <LAP> -u <UAP> -q <pcapfilename>`

jgalmeidah, 2017-05-02:
:o I understand.  So a process with ubertooth-rx (first find LAP and UAP, then the capture as above).

dominicgs, 2017-05-02:
Yes, that command assumes you have a target in mind, you can leave off the LAP and UAP to be less specific.  If you only provide the LAP it will calculate the UAP before trying to follow the piconet.

It should be noted that following a piconet is far from 100% accurate, so we do not receive all of the packet data.