greatscottgadgets/libbtbb #52: Wireshark Help

josch1, 2017-03-02:
This is not so much an issue as a question because  i'm fairly new to this, but its possible iv done something wrong.

I was following the instruction on gethub to capure BLE with Wireshark. https://github.com/greatscottgadgets/ubertooth/wiki/Capturing-BLE-in-Wireshark

But for some reason I only see a protocol for PPI ? 
I noticed in the command window it says "Control message unsupported", could that be the problem ?  
If so how can I fix it?
[image]

mikeryan, 2017-03-02:
See if following the instructions at the end of "Installing" helps (where it says Edit->Preferences):

https://lacklustre.net/bluetooth/wireshark.html

josch1, 2017-03-03:
Thank you!

 Im assuming the LE LL protocol is the Bluetooth Low Energy Link Layer?

[image]

One more question. For my senior project I have to capture Bluetooth packets from Fitbits, but i'm new to Wireshark, I don't know exactly what i'm looking at just yet. Know any threads or guides that could help me understand the packets details and bytes pane ?

straithe, 2021-02-17:
Do you still need assistance?

closed 2021-03-24 by straithe