greatscottgadgets/libbtbb #53: possible bug in btbb_winnow()

orgua, 2017-09-02:
hi Dominic and the rest of the team, 

i checked large parts of the code, optimized computation and found a hint to a possible bug: 

- bluetooth_piconet.c in function try_hop() 
   - `pn->winnowed = 0;` is commented out (THE hint)
   - the piconet-struct is given to btbb_winnow() on line 515 and 531
   - btbb_winnow starts a for-loop with the initial pn->winnowed value
   - on the first loop there could be an illegal operation happening if the value is 0:
```
last_index = pn->pattern_indices[pn->winnowed - 1];
last_channel = pn->pattern_channels[pn->winnowed - 1];
```
- shouldn't we add at least an assertion here? 
- it all depends on how btbb_process_packet() is called outside of the lib, it then calls try_hop() and so on

dominicgs, 2017-09-02:
That seems reasonable,or possibly return failure in some way when the incoming value is 0.

Have you been seeing issues caused by this?

orgua, 2017-09-04:
I have no hardware or dataset at hand atm, but will try the next days. i think the bug is not easy to observe. as i understand the code it will influence the afh-detection with undefined behaviour

straithe, 2021-05-04:
@orgua Are you still experiencing this issue? Or have you observed the bug again?

closed 2021-06-08 by straithe