greatscottgadgets/libbtbb #68: Fails to build with current Pythons: `setuptools.extern.packaging.version.InvalidVersion: Invalid version: ''`

barracuda156, 2024-07-16:
This has been working fine earlier, but now fails:
```
ports.org_macports_release_tarballs_ports_devel_libbtbb/libbtbb-devel/work/libbtbb-f0fe1768b080b1333a5357e8308d900e1273ec7f/lib/src/pcapng.c
[ 88%] Building C object lib/src/CMakeFiles/btbb.dir/pcapng-bt.c.o
cd /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_devel_libbtbb/libbtbb-devel/work/build/lib/src && /opt/local/bin/gcc-mp-14 -DRELEASE=\"2020-12-R1\" -DVERSION=\"1.0\" -Dbtbb_EXPORTS  -pipe -Os -DNDEBUG -I/opt/local/include -std=gnu90 -arch ppc -mmacosx-version-min=10.6 -fPIC   -Wall -MD -MT lib/src/CMakeFiles/btbb.dir/pcapng-bt.c.o -MF CMakeFiles/btbb.dir/pcapng-bt.c.o.d -o CMakeFiles/btbb.dir/pcapng-bt.c.o -c /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_devel_libbtbb/libbtbb-devel/work/libbtbb-f0fe1768b080b1333a5357e8308d900e1273ec7f/lib/src/pcapng-bt.c
Traceback (most recent call last):
  File "/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_devel_libbtbb/libbtbb-devel/work/build/python/pcaptools/setup.py", line 15, in <module>
    setup(
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/setuptools/_distutils/core.py", line 146, in setup
    _setup_distribution = dist = klass(attrs)
                                 ^^^^^^^^^^^^
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/setuptools/dist.py", line 318, in __init__
    self.metadata.version = self._normalize_version(self.metadata.version)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/setuptools/dist.py", line 354, in _normalize_version
    normalized = str(Version(version))
                     ^^^^^^^^^^^^^^^^
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/setuptools/_vendor/packaging/version.py", line 200, in __init__
    raise InvalidVersion(f"Invalid version: '{version}'")
setuptools.extern.packaging.version.InvalidVersion: Invalid version: ''
make[2]: *** [python/pcaptools/build/timestamp] Error 1
make[2]: Leaving directory `/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_devel_libbtbb/libbtbb-devel/work/build'
make[1]: *** [python/pcaptools/CMakeFiles/pcapdump.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Linking C shared library libbtbb.dylib
cd /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_devel_libbtbb/libbtbb-devel/work/build/lib/src && /opt/local/bin/cmake -E cmake_link_script CMakeFiles/btbb.dir/link.txt --verbose=ON
/opt/local/bin/gcc-mp-14 -pipe -Os -DNDEBUG -I/opt/local/include -std=gnu90 -arch ppc -mmacosx-version-min=10.6 -dynamiclib -Wl,-headerpad_max_install_names -L/opt/local/lib -Wl,-headerpad_max_install_names -compatibility_version 1.0.0 -current_version 1.0.0 -o libbtbb.1.0.dylib -install_name /opt/local/lib/libbtbb.1.dylib CMakeFiles/btbb.dir/bluetooth_packet.c.o CMakeFiles/btbb.dir/bluetooth_piconet.c.o CMakeFiles/btbb.dir/bluetooth_le_packet.c.o CMakeFiles/btbb.dir/companies.c.o CMakeFiles/btbb.dir/pcap.c.o CMakeFiles/btbb.dir/pcapng.c.o "CMakeFiles/btbb.dir/pcapng-bt.c.o"  -Wl,-rpath,/opt/local/lib
cd /opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_devel_libbtbb/libbtbb-devel/work/build/lib/src && /opt/local/bin/cmake -E cmake_symlink_library libbtbb.1.0.dylib libbtbb.1.dylib libbtbb.dylib
make[2]: Leaving directory `/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_devel_libbtbb/libbtbb-devel/work/build'
[100%] Built target btbb
make[1]: Leaving directory `/opt/local/var/macports/build/_opt_
```

ppss0, 2024-09-06:
I got this problem with pcapdump module installing too, and the lower version of setuptools DOES NOT work neither ! Any solutions?

hhlari, 2024-09-20:
I have same issue

kenballus, 2024-10-11:
It builds if you change the version to `'1.0'` in `setup.py`.

ukoda, 2025-01-20:
> It builds if you change the version to `'1.0'` in `setup.py`.

It is worth noting there is two `setup.py`files, I had to change both before it would build.